Larswad/uno2iec #5: Compiler problems/errors with VS2013

teki, 2015-02-11:
First is a warning:
"-std=gnu++0x" is not valid on Windows, qmake gives a warning, then during the build:
1>cl : Command line warning D9002: ignoring unknown option '-std=gnu++0x'

Then the build errors:
not, not_eq  and friends are macros. A few #include <iso646.h> is missing. (The best would be to remove the use of them all together, it makes to read the code hard.)
It builds with the following patch: http://pastebin.com/GhMZASZC

Larswad, 2015-02-11:
teki, thanks for sharing your findings and reporting it here.
Actually, I thought that I had made that c++0x switch dependant on platform so that the VS compilers would have the right switch. But obviously not (I have to recheck this).
I do use initializer lists on a couple of places that is c++0x additions, didn't that break under windows build if there is no explicit VS compiler switch for that?

And also, I thought that I had fixed the include for the (standard) alternate operators under Windows. Good findings, I will look at your patch and update.

Many thanks.

teki, 2015-02-11:
> I do use initializer lists on a couple of places that is c++0x additions, didn't that break under windows build if there is no explicit VS compiler switch for that?

It is not a problem for VS2013, it can be for earlier ones.

Larswad, 2015-04-01:
It should now conditonally compile correctly for both windows and Linux with the c++11 switch.
Please try it out.

Larswad, 2015-04-13:
Builds properly now. Closing.

closed 2015-04-13 by Larswad