Larswad/uno2iec #6: qextserialport has issues with Windows 8

teki, 2015-02-11:
qextserialport only iterates serial ports, but my UNO uses a USB-Serial driver, so it doesn't show up at all in settings.

A quick hack to make it work: http://pastebin.com/KVGgRGy1

(this patch is not 100% right, it removes serial com support, it should be easy to add it, qextserialport did support multiple classes, they just have to be iterated)

Larswad, 2015-02-11:
teki, This is also a good finding. This project originated from the Arduino UNO and was tested for that, so I wonder what has become different for Win8? Any project enumerating usb serial ports for arduino's should suffer under Win8 then, am I right? Or maybe I simply have to specify the device classes more explicitly like you do.

I have been thinking to throw out QExtSerialPort to see if the Qt 5.0+ native class QSerialPort would be a better replacement. Then I wouldn't need it as an external qt dependency.

teki, 2015-02-11:
Arduino IDE 1.6 ships with a new driver, maybe that changed things around.

teki, 2015-02-13:
Or maybe it is the board, do you have an R3 ? (with an atmel as ser->usb)

Larswad, 2015-02-13:
I will check once I get home. I have an arduino UNO clone from dealextreme so I'm not sure how this  board will map to the official revisions.

Larswad, 2015-04-01:
QextSerialPort has now been replaced with Qt's native QSerialPort, completely dropped the need for the QextSerialPort library. Please try it out and let me know if it works.

closed 2018-10-24 by Larswad