Larswad/uno2iec #7: Problem with putc on Centos 6.6

cb88, 2015-04-01:
So I build from Teki's fork as I was unable to get qextserialport to build at all... makefile error etc...

I have both gcc 4.4.7 and 4.8.2 but I'm not sure qtcreator is picking up on gcc 4.8.2 when I set it via the environment variables of the shell I run it from.

I also removed the Mac OS X ifdefs as for some reason it wouldn't build with those... I still have the following errors.

```
20:27:01: Running steps for project uno2iec-gui...
20:27:01: Configuration unchanged, skipping qmake step.
20:27:01: Starting: "/usr/bin/make" 
g++ -c -pipe -g -Wall -W -D_REENTRANT -fPIE -DCONSOLE_DEBUG -DQT_QML_DEBUG -DQT_DECLARATIVE_DEBUG -DQT_WIDGETS_LIB -DQT_GUI_LIB -DQT_SERIALPORT_LIB -DQT_CORE_LIB -I/usr/lib/qt5/mkspecs/linux-g++ -I../uno2iec-gui -isystem /usr/include/qt5 -isystem /usr/include/qt5/QtWidgets -isystem /usr/include/qt5/QtGui -isystem /usr/include/qt5/QtSerialPort -isystem /usr/include/qt5/QtCore -Idebug/.moc -Idebug/.ui -I. -o debug/.objnix/main.o ../uno2iec-gui/main.cpp
In file included from ../uno2iec-gui/interface.hpp:6,
                 from ../uno2iec-gui/mainwindow.hpp:10,
                 from ../uno2iec-gui/main.cpp:26:
../uno2iec-gui/filedriverbase.hpp:82:26: error: macro "putc" requires 2 arguments, but only 1 given
In file included from ../uno2iec-gui/interface.hpp:9,
                 from ../uno2iec-gui/mainwindow.hpp:10,
                 from ../uno2iec-gui/main.cpp:26:
../uno2iec-gui/m2idriver.hpp:42:18: error: macro "putc" requires 2 arguments, but only 1 given
In file included from ../uno2iec-gui/x00fs.hpp:4,
                 from ../uno2iec-gui/interface.hpp:10,
                 from ../uno2iec-gui/mainwindow.hpp:10,
                 from ../uno2iec-gui/main.cpp:26:
../uno2iec-gui/nativefs.hpp:51:18: error: macro "putc" requires 2 arguments, but only 1 given
In file included from ../uno2iec-gui/interface.hpp:6,
                 from ../uno2iec-gui/mainwindow.hpp:10,
                 from ../uno2iec-gui/main.cpp:26:
../uno2iec-gui/filedriverbase.hpp:82: error: 'putc' declared as a 'virtual' field
make: *** [debug/.objnix/main.o] Error 1
20:27:03: The process "/usr/bin/make" exited with code 2.
Error while building/deploying project uno2iec-gui (kit: Desktop)
When executing step 'Make'
20:27:03: Elapsed time: 00:03.
```

cb88, 2015-04-01:
From what I can tell putc is just an internal implementation so I just renamed it everywhere so gcc doesn't think it is trying to override the pubc from the standard library (I think)

I also uncommented the line below in the project file as it was required
```
QMAKE_CXXFLAGS += -std=c++0x 
```
21:10:37: Running steps for project uno2iec-gui...
21:10:37: Configuration unchanged, skipping qmake step.
21:10:37: Starting: "/usr/bin/make" 
g++ -c -pipe -std=c++0x -g -Wall -W -D_REENTRANT -fPIE -DCONSOLE_DEBUG -DQT_QML_DEBUG -DQT_DECLARATIVE_DEBUG -DQT_WIDGETS_LIB -DQT_GUI_LIB -DQT_SERIALPORT_LIB -DQT_CORE_LIB -I/usr/lib/qt5/mkspecs/linux-g++ -I../uno2iec-gui -isystem /usr/include/qt5 -isystem /usr/include/qt5/QtWidgets -isystem /usr/include/qt5/QtGui -isystem /usr/include/qt5/QtSerialPort -isystem /usr/include/qt5/QtCore -Idebug/.moc -Idebug/.ui -I. -o debug/.objnix/mainwindow.o ../uno2iec-gui/mainwindow.cpp
../uno2iec-gui/mainwindow.cpp: In member function 'void MainWindow::enumerateComPorts()':
../uno2iec-gui/mainwindow.cpp:277: error: no matching function for call to 'QSerialPortInfo::QSerialPortInfo(<brace-enclosed initializer list>)'
/usr/include/qt5/QtSerialPort/qserialportinfo.h:56: note: candidates are: QSerialPortInfo::QSerialPortInfo(const QSerialPortInfo&)
/usr/include/qt5/QtSerialPort/qserialportinfo.h:53: note:                 QSerialPortInfo::QSerialPortInfo()
make: *** [debug/.objnix/mainwindow.o] Error 1
21:10:41: The process "/usr/bin/make" exited with code 2.
Error while building/deploying project uno2iec-gui (kit: Desktop)
When executing step 'Make'
21:10:41: Elapsed time: 00:04.

Larswad, 2015-04-01:
Hi cb88!
Not sure if my cellphone mail arrived.
I'll write my answer properly here at github instead.
The problem with the c++11 (c++0x) problem: At the last commit I was compiling for microsoft compilers. Those do not recognise that flags (and the gnu compilers REQUIRES it).
I have fixed this problem now in the .pro file with conditional enabling depending on the platform.

I will push this as soon as I can!

The putc problem: Yes, good catch. It should be a gcc 4.8 problem or MacOS compiler specific, haven't seen it before. It seems like its what you say, they define the putc as a macro and that will definitely fail when I use it as a class member. It might work if I qualify with the class context on every call, but it might be better to rename it like you did . I will fix this, thanks!

The qextserialport: I am not sure why that doesn't build for you. Some other guy using MacOS did get it to work earlier.
However, I have a branch where I have tried to completely abandon qextserialport and use the QSerialPort (available in qt 5.3 and later). It seems to work just fine and I will push that branch as soon as possible. I just want to hook up the stuff and see that it works.

Larswad, 2015-04-01:
QextSerialPort has now been replaced with Qt's native QSerialPort, completely dropped the need for the QextSerialPort library. Please try it out and let me know if it works.

cb88, 2015-04-02:
So, the only compliation issue I am running into is 

```
#if !(defined(__APPLE__) || defined(_MSC_VER))
	QStringList list({ "PetMe64.ttf" , "PetMe2X.ttf" , "PetMe1282Y.ttf" });
#else
	// NOTE: QT5 for OSX has bug that won't let us use c++11 features. Special compile here until fixed.
	QStringList list;
	list << "PetMe64.ttf" << "PetMe2X.ttf" << "PetMe1282Y.ttf";
#endif
```
I haven't tested it yet other than starting the gui... I have a couple arduino nano clones from ebay I can test with tomorrow though.
Defaults to the top case which fails to compile.... Perhaps you should limit it to certain GCC versions? Althoughthe following might make it fail with clang... http://stackoverflow.com/questions/259248/how-do-i-test-the-current-version-of-gcc

cb88, 2015-04-02:
Also I only have a C128 to test with... no idea if this is a problem?

cb88, 2015-04-02:
I suspect that even though I have gcc 4.8 installed 4.4.7 is being used... qtcreator detects gcc 4.8 but I'm not sure how to tell it to use it.

Also I'm using the code that would be used for MacOS since I'm removing the c++0x code.... even though this is 32Bit Linux... I assume my compiler is just too old. It was defaulting to the c++0x code before I removed it though... which is what I would expect from your #if defines.
```
#else
    // NOTE: QT5 for OSX has bug that won't let us use c++11 features. Special compile here until fixed.
    QStringList list;
    list << "PetMe64.ttf" << "PetMe2X.ttf" << "PetMe1282Y.ttf";
#endif
```

Larswad, 2015-04-03:
Well, the c++0x (c++11) problem was that I STILL made some wrong platform conditional testing in the .pro file. I have commited those changes now.
Should enable you to build the c++11 stuff.

I built both on windows and linux and it should now work.
I have building on arduino 1.5.8 (and tested, it worked). I tested building (not running yet) using arduino 1.6.3 under linux, seems to work fine.

I think you need to check your cable, your PIN settings in the options dialog.
If you are using any kind of turboloader it will not work (pull out any eventual cartridge from the C64 or C128).

Larswad, 2018-10-24:
Closing this issue since no response and that it was resolved with the changes made.

closed 2018-10-24 by Larswad