Larswad/uno2iec #13: Save files other than .d64/fastloader implementation

n1cklarosa, 2017-01-04:
Hi There,

I love this product. I dont actually have an issue as my setup is working as the readme suggests (can run most single disk programs that have no built in fastloader using Arduino Nano). I was just curious to see if there was any plans to implement file saving from 3rd party programs (example - saving a file from a word processor) and/or support for support for some fastloaders?

Thanks for all the hard work!

kphair, 2017-01-05:
We regret to tell you Kevin passed away  on 4th December 2015,

Regards  Janine&Kevin, his parents.

Details:

http://notices.irishtimes.com/death/phair-kevin-j/44442552

cbmeeks, 2017-01-05:
I am so sorry to hear this.  Please pass my respect to along to his family.

Larswad, 2017-01-05:
I'm am deeply saddened to hear this, and my deepest sympathies for your loss.
Honestly, I wasn't  aware that Kevin was following this project or had commented in here earlier (as this message arrived in this thread for some reason). But from his github profile I can tell he was a dedicated vintage machine lover and software development. From the link, he was the same age as me.
Terrible message to receive.