Larswad/uno2iec #16: RP3 with Uno, error: Ending at char: 3

MarkBuechler, 2017-06-11:
I can't get past this particular issue. About one in 50 tries I manage to get a directory listing, but it usually fails with something like the following. I have tried an Uno and a Nano using /dev/ACM0 as well as /dev/AMA0 and /dev/ttyS0 (switching the two UARTS on RP3).

2017-06-11 18:53:02:917 S MAIN Now connected to Arduino.
2017-06-11 18:53:04:071 S R:MAIN CONNECTED, READY FOR IEC DATA WITH CBM AS DEV 8.
2017-06-11 18:53:04:073 I R:MAIN IEC pins: ATN:5 CLK:4 DATA:3 RST:7 SRQIN:2
2017-06-11 18:53:04:074 I R:MAIN Arduino time set to: 2017-06-11.18:53:02
2017-06-11 18:53:07:178 E R:IFACE ATNCMD: IEC_ERROR!
2017-06-11 18:53:12:961 I R:IFACE ATN code:240 cmd: * (len: 1) retATN: 1
2017-06-11 18:53:12:962 I IFACE processOpenCommand, cmd: *
2017-06-11 18:53:12:963 W IFACE File * not found. Returning FNF to CBM.
2017-06-11 18:53:12:965 E IFACE Open ReadPRG Response code: 0
2017-06-11 18:53:13:665 I R:IFACE ATN code:96 cmd:  (len: 0) retATN: 3
2017-06-11 18:53:16:274 I R:IFACE ATN code:240 cmd: $ (len: 1) retATN: 1
2017-06-11 18:53:16:650 I IFACE processOpenCommand, cmd: $
2017-06-11 18:53:16:651 I IFACE Producing directory listing for FS: ""...
2017-06-11 18:53:16:693 I NATIVEFS Listing 20 entrie(s) to CBM.
2017-06-11 18:53:16:695 S IFACE Directory listing ok (21 lines). Ready waiting for line requests from arduino.
2017-06-11 18:53:16:697 S IFACE Open ReadPRG Response code: 3
2017-06-11 18:53:16:824 I R:IFACE ATN code:96 cmd:  (len: 0) retATN: 3
2017-06-11 18:53:18:179 E R:IFACE Ending at char: 3.
2017-06-11 18:53:19:738 E R:IFACE L
2017-06-11 18:53:20:094 I R:IFACE ATN code:224 cmd:  (len: 0) retATN: 1

MarkBuechler, 2017-06-13:
This has been resolved. It's timing related. I was using a remote X server which slowed down the main loop just enough. I'm guessing your 1541 code must run in the main loop instead of a separate thread.

Larswad, 2018-10-24:
Great, then closing this.

closed 2018-10-24 by Larswad