Larswad/uno2iec #18: Error: /MP: No such file or directory with Windows 10 QT Creator

fl3tching101, 2018-03-19:
I get the error: Error: /MP: No such file or directory when I try to compile the rpi2iec.pro in QT Creator. Have I done something wrong or is this an issue? Thanks.

str3l0ck, 2018-10-24:
I get the same issue on Windows XP as well.

`22:14:23: Running steps for project rpi2iec...
22:14:23: Configuration unchanged, skipping qmake step.
22:14:23: Starting: "C:\Qt\Qt5.5.1\Tools\mingw492_32\bin\mingw32-make.exe" 
C:/Qt/Qt5.5.1/Tools/mingw492_32/bin/mingw32-make -f Makefile.Release
mingw32-make[1]: Entering directory 'C:/uno2iec-0.5.0.0'
C:\Qt\Qt5.5.1\5.5\mingw492_32\bin\uic.exe mainwindow.ui -o release\.ui\ui_mainwindow.h
C:\Qt\Qt5.5.1\5.5\mingw492_32\bin\uic.exe aboutdialog.ui -o release\.ui\ui_aboutdialog.h
C:\Qt\Qt5.5.1\5.5\mingw492_32\bin\uic.exe settingsdialog.ui -o release\.ui\ui_settingsdialog.h
C:\Qt\Qt5.5.1\5.5\mingw492_32\bin\uic.exe logfiltersetup.ui -o release\.ui\ui_logfiltersetup.h
C:\Qt\Qt5.5.1\5.5\mingw492_32\bin\uic.exe mountspecificfile.ui -o release\.ui\ui_mountspecificfile.h
g++ -c -pipe -fno-keep-inline-dllexport /MP -O2 -frtti -Wall -Wextra -fexceptions -mthreads -DUNICODE -DCONSOLE_DEBUG -DQT_NO_DEBUG -DQT_WIDGETS_LIB -DQT_GUI_LIB -DQT_SERIALPORT_LIB -DQT_CORE_LIB -DQT_NEEDS_QMAIN -I. -I..\Qt\Qt5.5.1\5.5\mingw492_32\include -I..\Qt\Qt5.5.1\5.5\mingw492_32\include\QtWidgets -I..\Qt\Qt5.5.1\5.5\mingw492_32\include\QtGui -I..\Qt\Qt5.5.1\5.5\mingw492_32\include\QtANGLE -I..\Qt\Qt5.5.1\5.5\mingw492_32\include\QtSerialPort -I..\Qt\Qt5.5.1\5.5\mingw492_32\include\QtCore -Irelease\.moc -Irelease\.ui -I..\Qt\Qt5.5.1\5.5\mingw492_32\mkspecs\win32-g++  -o release\.objwin\main.o main.cpp
g++: error: /MP: No such file or directory
Makefile.Release:1694: recipe for target 'release/.objwin/main.o' failed
mingw32-make[1]: Leaving directory 'C:/uno2iec-0.5.0.0'
makefile:34: recipe for target 'release' failed
mingw32-make[1]: *** [release/.objwin/main.o] Error 1
mingw32-make: *** [release] Error 2
22:14:24: The process "C:\Qt\Qt5.5.1\Tools\mingw492_32\bin\mingw32-make.exe" exited with code 2.
Error while building/deploying project rpi2iec (kit: Desktop Qt 5.5.1 MinGW 32bit)
When executing step "Make"
22:14:24: Elapsed time: 00:01.`

Larswad, 2018-10-24:
str3l0ck: Somehow the /MP switch is added even if you compile with mingw. It is meant for the Microsoft msvc builds. Mingw doesn't understand that switch. It is for multicpu build under msvc, instead mingw make uses - j[numthreads]. So, try in the pro file to remove the line that adds the /MP switch to the compiler settings.

str3l0ck, 2018-10-27:
Commenting out line 19 gives the following:
01:31:16: Running steps for project rpi2iec...
01:31:16: Configuration unchanged, skipping qmake step.
01:31:16: Starting: "C:\Qt\Qt5.5.1\Tools\mingw492_32\bin\mingw32-make.exe" 
C:/Qt/Qt5.5.1/Tools/mingw492_32/bin/mingw32-make -f Makefile.Release
mingw32-make[1]: Entering directory 'C:/uno2iec-0.5.0.0'
C:\Qt\Qt5.5.1\5.5\mingw492_32\bin\uic.exe mainwindow.ui -o release\.ui\ui_mainwindow.h
C:\Qt\Qt5.5.1\5.5\mingw492_32\bin\uic.exe aboutdialog.ui -o release\.ui\ui_aboutdialog.h
C:\Qt\Qt5.5.1\5.5\mingw492_32\bin\uic.exe settingsdialog.ui -o release\.ui\ui_settingsdialog.h
C:\Qt\Qt5.5.1\5.5\mingw492_32\bin\uic.exe logfiltersetup.ui -o release\.ui\ui_logfiltersetup.h
C:\Qt\Qt5.5.1\5.5\mingw492_32\bin\uic.exe mountspecificfile.ui -o release\.ui\ui_mountspecificfile.h
g++ -c -pipe -fno-keep-inline-dllexport -O2 -frtti -Wall -Wextra -fexceptions -mthreads -DUNICODE -DCONSOLE_DEBUG -DQT_NO_DEBUG -DQT_WIDGETS_LIB -DQT_GUI_LIB -DQT_SERIALPORT_LIB -DQT_CORE_LIB -DQT_NEEDS_QMAIN -I. -I..\Qt\Qt5.5.1\5.5\mingw492_32\include -I..\Qt\Qt5.5.1\5.5\mingw492_32\include\QtWidgets -I..\Qt\Qt5.5.1\5.5\mingw492_32\include\QtGui -I..\Qt\Qt5.5.1\5.5\mingw492_32\include\QtANGLE -I..\Qt\Qt5.5.1\5.5\mingw492_32\include\QtSerialPort -I..\Qt\Qt5.5.1\5.5\mingw492_32\include\QtCore -Irelease\.moc -Irelease\.ui -I..\Qt\Qt5.5.1\5.5\mingw492_32\mkspecs\win32-g++  -o release\.objwin\main.o main.cpp
In file included from interface.hpp:7:0,
                 from mainwindow.hpp:10,
                 from main.cpp:26:
d64driver.hpp: In member function 'virtual const QStringList& D64::extension() const':
d64driver.hpp:43:32: warning: extended initializer lists only available with -std=c++11 or -std=gnu++11
   static const QStringList ext({ "D64" });
                                ^
d64driver.hpp:43:41: warning: extended initializer lists only available with -std=c++11 or -std=gnu++11
   static const QStringList ext({ "D64" });
                                         ^
d64driver.hpp:43:41: warning: extended initializer lists only available with -std=c++11 or -std=gnu++11
d64driver.hpp:43:41: warning: extended initializer lists only available with -std=c++11 or -std=gnu++11
d64driver.hpp:43:41: error: call of overloaded 'QStringList(<brace-enclosed initializer list>)' is ambiguous
d64driver.hpp:43:41: note: candidates are:
In file included from ..\Qt\Qt5.5.1\5.5\mingw492_32\include/QtCore/qlist.h:1026:0,
                 from ..\Qt\Qt5.5.1\5.5\mingw492_32\include/QtCore/qobject.h:43,
                 from ..\Qt\Qt5.5.1\5.5\mingw492_32\include/QtCore/qcoreapplication.h:40,
                 from ..\Qt\Qt5.5.1\5.5\mingw492_32\include\QtWidgets/qapplication.h:37,
                 from ..\Qt\Qt5.5.1\5.5\mingw492_32\include\QtWidgets/QApplication:1,
                 from main.cpp:17:
..\Qt\Qt5.5.1\5.5\mingw492_32\include/QtCore/qstringlist.h:98:21: note: QStringList::QStringList(const QString&)
     inline explicit QStringList(const QString &i) { append(i); }
                     ^
..\Qt\Qt5.5.1\5.5\mingw492_32\include/QtCore/qstringlist.h:93:7: note: QStringList::QStringList(const QStringList&)
 class QStringList : public QList<QString>
       ^
In file included from interface.hpp:8:0,
                 from mainwindow.hpp:10,
                 from main.cpp:26:
t64driver.hpp: In member function 'virtual const QStringList& T64::extension() const':
t64driver.hpp:17:32: warning: extended initializer lists only available with -std=c++11 or -std=gnu++11
   static const QStringList ext({ "T64" });
                                ^
t64driver.hpp:17:41: warning: extended initializer lists only available with -std=c++11 or -std=gnu++11
   static const QStringList ext({ "T64" });
                                         ^
t64driver.hpp:17:41: warning: extended initializer lists only available with -std=c++11 or -std=gnu++11
t64driver.hpp:17:41: warning: extended initializer lists only available with -std=c++11 or -std=gnu++11
t64driver.hpp:17:41: error: call of overloaded 'QStringList(<brace-enclosed initializer list>)' is ambiguous
t64driver.hpp:17:41: note: candidates are:
In file included from ..\Qt\Qt5.5.1\5.5\mingw492_32\include/QtCore/qlist.h:1026:0,
                 from ..\Qt\Qt5.5.1\5.5\mingw492_32\include/QtCore/qobject.h:43,
                 from ..\Qt\Qt5.5.1\5.5\mingw492_32\include/QtCore/qcoreapplication.h:40,
                 from ..\Qt\Qt5.5.1\5.5\mingw492_32\include\QtWidgets/qapplication.h:37,
                 from ..\Qt\Qt5.5.1\5.5\mingw492_32\include\QtWidgets/QApplication:1,
                 from main.cpp:17:
..\Qt\Qt5.5.1\5.5\mingw492_32\include/QtCore/qstringlist.h:98:21: note: QStringList::QStringList(const QString&)
     inline explicit QStringList(const QString &i) { append(i); }
                     ^
..\Qt\Qt5.5.1\5.5\mingw492_32\include/QtCore/qstringlist.h:93:7: note: QStringList::QStringList(const QStringList&)
 class QStringList : public QList<QString>
       ^
In file included from interface.hpp:9:0,
                 from mainwindow.hpp:10,
                 from main.cpp:26:
m2idriver.hpp: In member function 'virtual const QStringList& M2I::extension() const':
m2idriver.hpp:15:32: warning: extended initializer lists only available with -std=c++11 or -std=gnu++11
   static const QStringList ext({ "M2I" });
[... 106 lines cut ...]
..\Qt\Qt5.5.1\5.5\mingw492_32\include/QtCore/qstringlist.h:99:12: note: QStringList::QStringList(const QList<QString>&)
[... cut ...]
Makefile.Release:1694: recipe for target 'release/.objwin/main.o' failed
mingw32-make[1]: *** [release/.objwin/main.o] Error 1
mingw32-make[1]: Leaving directory 'C:/uno2iec-0.5.0.0'
mingw32-make: *** [release] Error 2
makefile:34: recipe for target 'release' failed
01:31:19: The process "C:\Qt\Qt5.5.1\Tools\mingw492_32\bin\mingw32-make.exe" exited with code 2.
Error while building/deploying project rpi2iec (kit: Desktop Qt 5.5.1 MinGW 32bit)
When executing step "Make"
01:31:19: Elapsed time: 00:03.

str3l0ck, 2018-11-15:
Just wanted to let you know I finally got this working and it's pretty awesome! I still have the /MP issue on the 0.5.0.0 version but the 0.4.0.0 compiles and runs without issue provided I use the right version of QT. I got it to work with 5.3.2 x86 with the mingw compiler instead of msvc. Apparently msvc2012 produces unusable binaries on Windows XP so none of the versions of QT using that compiler would install. Also I had to use a version of Arduino IDE older than 1.5.7 since there is some code that was changed in the Arduino libraries in newer versions that don't work with your code. Thanks for the awesome project, now to finally play some games!