Larswad/uno2iec #19: Reconnect loop

r1me, 2018-07-23:
Hardware: Arduino UNO R3 (ATMEGA328P-PU board with ATMEGA16U2), PC with Windows 7 x64
Software: Qt 5.11.1 with MinGW32 compiler, Arduino IDE 1.8.5

I've double checked connections
[image]
but it goes into an infinite reconnect loop:
[image]
I've tried compiling sketch with older IDE version - Arduino 1.5.2 but it made no difference.
Can UNO2IEC be used with Windows 7 x64 and Arduino UNO R3 ? Can someone confirm this setup is possible ?
How to debug/log messages between PC and UNO if port is occupied for communication purpose ?

r1me, 2018-07-23:
I've got it working with those files: https://ccomodore64.blogspot.com/2018/04/uno2iec-czyli-emulator-stacji-dyskow.html

closed 2018-07-23 by r1me

Larswad, 2018-07-23:
Hello r1me, sorry for coming in late here. It's vacation times and somewhat offline for the moment. I'm glad thing got straightened out, but I'm kind of curious though. What was more specifically the difference? I mean, the head commit SHOULD work for the Uno, both the Windows side and the arduino side. 
If there are any problems or something missing in my documentation, please let me know or file an issue. 
Nevertheless, for a better experience try my port of the sd2iec project for the mega2560, in that you have all the features of turbo and support for saving etc. Only limitation is that it for the moment is limited to sd cards (like sd2iec is).

r1me, 2018-07-23:
Hey, it is not caused by the Qt or Arduino IDE version mismatch or QtSerialPort incompatibility etc. but the Arduino sketch.
I've just confirmed it by building and deploying Qt app from master branch, and then tried Arduino sketch from master and r0.5. Results are: sketch in release 0.5 of your project works just fine. Sketch from master branch causes the reconnection issue. 
I've noticed the difference in default baud rate between master and r0.5, r0.5 defaults to 57600. However, changing DEFAULT_BAUD_RATE to 57600 with master branch sketch, doesn't solve the issue. If you have any ideas where to apply patch, I'm willing to test it.

Larswad, 2018-07-23:
Thanks so much for your info.! Never expected the master to be flawed. But I will test thing and correct once I have the stuff rigged up again.
It's been a while.

kphair, 2018-08-08:
We regret to tell you Kevin passed away  on 4th December 2015,

Regards  Janine&Kevin, his parents.

Details:

http://notices.irishtimes.com/death/phair-kevin-j/44442552