Larswad/uno2iec #21: Loading files after main program not possible?

creepebrine, 2019-05-04:
I wanted to load a game named Planet X2 on my C64 via uno2iec, because it was the most affordable FDD emulator in my area. After loading the main program (the game itself) I was able to start it. After the main menu it kinda freezes, because it wants to load maps but seems not to be able to do that.
Here's the error stuff I got from the log:

2019-05-04 19:47:59:040 I IFACE processOpenCommand, cmd: M-W
2019-05-04 19:47:59:040 W DOSCMD M-W 0x500:32
2019-05-04 19:47:59:040 S IFACE CmdChannel_Response code: 0 = '00,OK'
2019-05-04 19:47:59:040 I IFACE processOpenCommand, cmd: M-W ?? ?7 ? ????< `C
???????8FHU
2019-05-04 19:47:59:056 W DOSCMD M-W 0x520:32
2019-05-04 19:47:59:056 S IFACE CmdChannel_Response code: 0 = '00,OK'
2019-05-04 19:47:59:056 I IFACE processOpenCommand, cmd: M-E
2019-05-04 19:47:59:056 E IFACE CmdChannel_Response code: 98 = '98,NOT IMPLEMENTED'
2019-05-04 19:47:59:228 I IFACE processOpenCommand, cmd: M-R; 
2019-05-04 19:47:59:228 E IFACE CmdChannel_Response code: 98 = '98,NOT IMPLEMENTED'
2019-05-04 19:47:59:228 I IFACE processOpenCommand, cmd: 
2019-05-04 19:47:59:228 I IFACE CmdChannel Status Response code: 98 = '98,NOT IMPLEMENTED'

I also tried to load a SID file into a C64 SID player and it also won't load, causing the program to freeze.
Thanks in advance.
Greetings from Germany!

Larswad, 2019-05-04:
Hello! 
I can see that the c64 is trying to read or write to the drive ram memory. That is a sign of that it wants to upload a turbo loader to the drive. 
This is not supported by uno2iec.
You can load some other file or program that doesn't use turbos, but anything that will try to use turbos will fail. 

I have made a port of the sd2iec for the arduino mega2560. That will handle the most common turbos well.

However, the absolutely best option today is to use the 1541 raspberry pi 1541pi, it fully implements all aspects of the drive.

You will however need a levelconverter for the signals from 5v to 3v.
They can be cheeply available in most internet electronic stores. 

Best regards, 
Lars

⁣Skickat från TypeApp ​

creepebrine, 2019-05-05:
I tried the Pi1541 with my Pi 3 before I made myself an uno2iec. I used a level converter and double checked the cable connections and always got a "Device not found" error with multiple ROMs and drive numbers. It seems to notice when I turn my C64 on though.

creepebrine, 2019-05-05:
UPDATE: I just found a version that does not use a fast loader to load the maps. But there is another problem: It seems like "loading on channel 4 is not yet implemented" (what I can remember the log said).
Will this function be added? The maps are SEQ files as far as I know.

Larswad, 2019-09-24:
Hi creepebrine!
It's been quite a while now since I did anything on this project.
It feels less motivated now as I both have the sd2iec port to the Mega2560  AND the PI1541 project (who both are much better than my uno2iec project).

Well, never say never, but I am not so motivated to add features to this project as it is now becoming quite obsolete.
I would guess it is exactly that, the SEQ files are loaded on that channel, and I never implemented everything.

You could probably give it a shot yourself if you're into Qt and arduino, it shouldn't be that hard to add some support for SEQ files.

/Lars