Larswad/uno2iec #22: Teensy2.0++ with teensyduino giving a protocol version mismatch ?

QuinTeknoLife, 2020-02-07:
So since i had a teensy2.0++ just laying around and teensy boards apparently run arduino code I decided to give this a shot. Compiling everything went fine and the board accepted the code perfectly but during runtime I get a protocol ver mismatch (apparently the board sends 0 but the program expects 2).

Is there any way to get this program to work with teensy?

Larswad, 2020-02-09:
Hi, I don't have a Teensy of my own, so I'll be unable to test.
The most probably thing I can imagine is that the Teensy side is the one that formats the string in a way that doesn't work in the same way as for the arduinos.
Maybe try this, change line 126 of uno2iec.ino to:

`sprintf_P(tempBuffer, connectionString, unsigned(CURRENT_UNO2IEC_PROTOCOL_VERSION));`

Please try that and see what happens.

QuinTeknoLife, 2020-02-11:
I have tried the change of code you suggested, but it did not cause for any changes in behavior and still gives the same message.

The string that the teensy is putting out apparently is this one
connect_arduino:2

[image]

QuinTeknoLife, 2020-02-18:
So I tried to manually hard-set the protocol ver in QtStudio/mainwindow to 2, resulting in a connected arduino but a bunch of ATNCMD IEC_ERRORs

niclashoyer, 2020-04-19:
@QuinTeknoLife @Larswad 
I got this working yesterday on a Teensy 2.0 that I used as a cheap ZoomFloppy before. Might be easily portable to the Teensy 2.0++. I had several timing issues, because the internal serial/USB connection is really fast and always at 12MB/s speed, more info [on the teensyduino page](https://www.pjrc.com/teensy/td_serial.html). I had to insert several while loops that wait for data in the read buffer.

Also I had issues with `\r` line endings on Linux, I had to replace them with `\n`.

All changes including hex file for Teensy 2.0 is in #24.