Larswad/uno2iec #26: ESP32 compatibility

Zibri, 2020-05-20:
Do you think this will work on an ESP32?
It would be great to have it and using wifi too :D or an sdcard cooonected to the ESP32

Larswad, 2020-05-20:
It is more than enough capable from performance point of view. It would however need voltage converters for the IEC pins because the esp32 is 3.3V and the open collector bus on the c64 is 5V.
Due it its higher performance I would be surprised if some timings must be changed. 

But yeah, with some efforts it would work. But since it actually features an SD card it would make more sense it port the sd2iec project. I did that with the mega2560 board. So I guess it could be done with this one too.

Zibri, 2020-05-28:
what I was thinking is:

real C64 >> tape and iec port to esp32 >>wifi >>pc   and
real C64 >> tape and iec port to esp32 >>sdcard

in order to use TAP and D64/G64 hosted on pc or sdcard

Zibri, 2020-12-17:
any news about this?

Larswad, 2021-01-19:
I have not ordered one. 
It was a while since I worked on the project. 
I am not really sure the best place to buy one. 
It does feel however as a little overkill for this particular project. 
I am thinking about it. 
BR
Lars

⁣Skickat från TypeApp ​

Zibri, 2023-05-20:
> It is more than enough capable from performance point of view. It would however need voltage converters for the IEC pins because the esp32 is 3.3V and the open collector bus on the c64 is 5V. Due it [...]

No level converters are needed. ESP32 is 5v tolerant.
Check here: https://github.com/idolpx/meatloaf