Larswad/uno2iec #27: Which max7219 library do you use?

pottendo, 2020-10-09:
hi, 
successfull ran uno2iec (with its limits) today - great work!
Wanted to activate the 7219 LED matrix - which arduino library do you use? I've tried several ("LedControl", minimalistic max7219, etc.) but non compiled, either because the header <max7219,h> is even missing or the defined classes are different.

thanks  for any input.
pottendo

PS: is anybody trying to improve the compatibility, these days? I've also stumbled of the strange-char crash, etc.

Larswad, 2020-10-09:
Hi Pottendo! 

Thanks for your words and interest!

Hmm, I think that you can use my own cloned library of the max 7219. Only very slight modifications. Look in my other project. 
If that was the question? I mean, I don't think there were other dependencies than that. 

I promise myself to do something about it. 
But It always ends up with other things to do. Felt a little bit unmotivated when the pi1541 does real emulation as close as possible. 

If you have arduino mega 2560 you can try out my port of the sd2iec, works with turbos and all that. 

Let me know if you have more questions. 
BR
Lars

⁣Skickat från TypeApp ​

pottendo, 2020-10-09:
hi,
thanks for the fast response - no I don't have a mega - I just have the Uno and several esp's (8266, 32). However, megas are also affordable... ;-)
I'd just like to copy .d64 to real floppies... so if the character issue is fixed, I think I could live with that (or buy a mega).

ad the 7219 lib: I have to double check - it failed to compile in the first place because of the missing max7219.h header, so I thought I install one... but there are many, and most even have no max721.h at all but some other lib-specific header.
I'll look into you projects.

thanx, pottendo

pottendo, 2020-10-09:
hi,
with your library the code compiles (almost).
Had to change:
  pMax->resetScrollText_p(myText); 
to 
  pMax->resetScrollText(myText);
in two locations.

Cool font! not really easy to read, but cool! ;-)
all the best, pottendo

closed 2020-10-09 by pottendo

Larswad, 2020-10-09:
Hi, glad to hear!
It's been a while since I fiddled with the max. and to be honest I'm a little surprised that you got it working with so few changes. 

The font, it is actually the font from the c64 game "parallax" (my favorite from those days, it's so cool) . You can exchange it with any c64 font if you know how very simple they are formatted. Eight bytes per character. 
They are in petscii order. 
I think there probably exist som zipped collections of c64 fonts somewhere on the net. Just remember that the first two bytes MIGHT be the loading address in the c64 memory, they have to (I think) be removed, unless I actually thought of that. 

Yet still, if you want to convert your floppies, your best bet is my port of the sd2iec. 
If you get the save command to work in uno2iec, just go ahead (I might have had that working at some time). 

⁣Skickat från TypeApp ​

Larswad, 2020-10-09:
Hi again, 
BTW, if you have the time, it would be very nice to see a picture of your little build. It doesn't have to look or anything, it's just that I really enjoy seeing that people still actually goes that long way these days and give uno2iec a fair shot. 

Great job getting it all build able, I mean, qt has changed, the arduino studio and build suite has changed since so I think it's surprising almost that it builds (although I realize you probably have fought a bit with details). 
BR, 
Lars

⁣Skickat från TypeApp ​