Larswad/uno2iec #28: Not able to get connection to uno2iec from C64

FHoevi, 2021-01-11:
Hi there,
I installed Qt on a WIN 10 box, and built the rp2iec.exe from source. Then I took an Arduino Uno Rev 3 SMD and successfully uploaded the sketch. Next, I created a cable doing the following connection of Arduino (A) pins to a female 6-pin DIN connector (D) according to the instruction [](https://www.c64-wiki.com/wiki/Serial_Port)

```
A3 -> D5 DATA,
A4 -> D4 CLOCK,
A5 -> D3 ATN,
A6 -> D1 SRQ,
A7 -> D6 RESET,
AGND -> D2 GND.
```

After connecing anything properly I started the executable and got the following messages in the log window:

```
2021-01-11 20:37:35:311 S MAIN Application Started, using port COM10 @ 115200
2021-01-11 20:37:35:323 S MAIN Application Initialized.
2021-01-11 20:37:36:314 S MAIN Now connected to Arduino.
2021-01-11 20:37:36:817 S R:c%c CONNECTED, READY FOR IEC DATA WITH CBM AS DEV 8.
2021-01-11 20:37:36:822 I R:c%c IEC pins: ATN:5 CLK:4 DATA:3 RST:7 SRQIN:2
2021-01-11 20:37:36:825 I R:c%c Arduino time set to: 2021-01-11.20:37:36
```

So far, so good. Next step was to mount a d64-file on my WIN 10 host box which led to:

```
2021-01-11 20:46:24:691 I IFACE processOpenCommand, cmd: Summer_Games_I+II_(HF)_Disk_1.d64
2021-01-11 20:46:24:692 I IFACE Trying image mount using driver: D64|D64
2021-01-11 20:46:24:696 S IFACE Mount OK of image: Image: Summer_Games_I+II_(HF)_Disk_1.d64
```

with corresponding information to be found in the "Current T64 / D64 Internal Directory Listing" section.

On the C64 I typed the usual `LOAD"$",8`.

The reply was:
```
SEARCHING FOR $
? DEVICE NOT PRESENT ERROR.
```

What is going wrong? Thanks in advance for any kind of suggestions.

Larswad, 2021-01-11:
Hi FHOevi! 
Well if you're lucky you just might have done correct on the physical wiring, I mean, no soldering mistakes or alike. 
But I hope that your are aware that your must also select the very same pins in the unoo2iec (Qt) application on the configuration menu. Unless doing so, you might just have a wrong software configuration. Open up the menu and select the proper pins in the menu. When the PC application starts and connects to the arduino it sends those pins you have selected to the arduino, then arduino then know on what pins to communicate with the c64. 
To me, it sounds as if you have simply forgotten to do jst that. Please report back.

Larswad, 2021-01-11:
Oh by the way, the srqin pin is not used by uno2iec, it was a reservation in case being prepared for 1571 operation in future, but was never supported.

FHoevi, 2021-01-12:
Hi Larswad,
thanks for replying immediately.

Well, I already checked hopefully anything necessary before opening the issue. In settings of the host software I have the following entries:

```
COM Port COM10
Serial Speed 115200
Device number 8
Reset pin (optional) 7
Clock Pin 4
ATN Pin 5
Data Pin 3
```

The GUI reads back (like already reported above):

`IEC pins: ATN:5 CLK:4 DATA:3 RST:7 SRQIN:2`

But IEC pins are the pins for the 6-pin DIN connector (the "D" side from above), right? If so, those would not correspond to the specs in, e.g. [](https://www.c64-wiki.com/wiki/Serial_Port) where you could find the pin layout as mentioned in my original post.

Best

Larswad, 2021-01-12:
Hi again, 
What I don't understand is that in your first wiring table you wrote in the first post, you mention that you have the data connected to digital #5, and the ATN at #3. Yet still you say that in software configuration, you specify data on #3 and ATN on #5. If that really is the case, then it's obviously wrong. 

Am I misunderstanding you here somehow? 

BR, 
Lars

⁣Skickat från TypeApp ​

FHoevi, 2021-01-12:
Well, no, it's not **D**igital 5 on the Arduino what I meant but **A**(rduino)3 maps to **D**(IN connector)5. So, I think I did it right.

Larswad, 2021-01-12:
Ok, then I understand you. Sorry for that.

Then there is also this thing that is easy to mix up. 
Originally you have to consider the pins to be on the connector side of the pins, not the c64 port.
However, there is a double possibility of confusion, when you solder, since you actually do that on the back side of the connector, the pins will in fact (still) map to the way they are when you look at the c64 port.

I know this is maybe too simple mistakes and probably you have thought things correct, but I do ask because there is not much that can go wrong once you have soldered the connector and attached it.

Please try also see if that filename that you used could cause any problems, I saw there was some + character in it, don't know if that can cause something strange.

FHoevi, 2021-01-12:
Well, as I wrote above, I used a female connector, mainly for being able to use a standard serial cable. Thus I had to do the wiring exactly as outlined here https://www.c64-wiki.com/wiki/Serial_Port, right? When I look at my connector as if I would be plugging in that cable, then the wiring is done like on the picture from that website.

OK, will rename the file and try again. But to my feeling the error message on the C64 side means that the connection C64 -> Arduino does not work whereas the connection PC -> Arduino is working well.

FHoevi, 2021-01-12:
Renaming the .d64 file did not have an effect, still not working with the same error message from C64.

BTW, the C64 in use is able to communicate with two different 1541s.

Larswad, 2021-01-12:
Yes, remember, that picture at c64 wiki shows the the female c64 side. Here is a picture that maybe makes things clear:

https://allpinouts.org/pinouts/connectors/serial/commodore-c64-c16-c116-4-serial/
(both rotated compared to the c64 wiki picture).

FHoevi, 2021-01-12:
Well, that was exactly the picture I had as well and I went ahead accordingly. **But** while checking anything again, this time with a multimeter, I realized that for the female connector in use the pinout for pins 3 (ATN) and 6 (RST) was opposite to what I thought I checked right in the beginning after receiving the connector. Thus, in the GUI I exchanged Arduino pins 5 and 7 and now it works.

Thanks for the discussion, remote troubleshooting, and pushing me to check over and over again to be able to see that my own isufficiency was the reason for it not working. What else should it be?

closed 2021-01-12 by FHoevi

Larswad, 2021-01-12:
Great to hear! 

It is very easy to do such mistakes. 
If you get hold of an arduino mega 2560, I have made a port of the sd2iec that works for that. It was a matter of some timing changes and such. 
That one supports turbos and such, even though it doesn't use a GUI host like uno2iec. 
I actually started some work to try this get the sd2iec support the qt GUI in uno2iec, but didn't finish. Maybe I'll take up on that som day. 

What brings my motivation down is that the pi1541 already does perfect emulation. It is I feel the best project out there by far. 

I am glad however if this old uno2iec still makes someone happy. 

Best regards 
Lars

⁣Skickat från TypeApp ​

FHoevi, 2021-01-12:
Well, yes, let me have a look at the Mega 2560 port, I have some here on my desk. Where can I find it?

I certainly know about many RPi based CBM emulators, but in this spirit you could run off old games from the RPi directly, that's even more convenient (RetroPi or so). My main intention was to get the original C64 up and running again together with the original hardware including a heavily flickering tube monitor. The R/W heads of my two 1541s need some adjustment and some cleaning which is only possible if I get the adjustment software onto the C64 in another way. I have a couple of Arduinos here so why not checking out uno2iec then? At least to be able to show a teenager or almost grown-up how gaming felt like back in the 80s when I was at his current age...

Larswad, 2021-01-12:
Good idea yes to show kids the old good times and games. Yes I understand, you needed some way to get those calibration programs running. 

What I meant with the rpi (pi 1541) was not a c64 emulator, it is a down to the cycle correct, bare-metal 1541/1571/1581 emulator, where you solder a shield (board) with voltage divider, solder one or two iec connectors and some other components, even a buzzer (for drive sounds) and a small lcd for file view (using button switches) . What you get is perfect drive emulation. It is a fantastic project they made. 

About c64 emulators and the pi, the absolutely best one I can recommend is the BMC64, because it is totally bare metal (not Linux or other OS) and it is able to set the exact frame rate of the c64 (e.g. 50.12 Hz for PAL). 
I would say it is as close as you ever can come. 
BR, 
Lars

⁣Skickat från TypeApp ​

anatolbaskak, 2023-10-17:
Hi, may I join in?
I'm struggling with NANO2IEC device (a variant of UNO2IEC, obviously). I seem to have checked everything and I'm still getting the "Device not present" error. I *think* I've checked it when I bought it, 2 years ago, and it worked, but now it doesn't, on two C64s. I did check connections between solder joints on the board and socket receptacles. Any suggestions?
Thanks in advance

[image]
[image]
[image]
[image]
[image]
[image]

anatolbaskak, 2023-10-18:
Hello Lars, and many thanks for prompt reply.
I used the Polish-language instruction page (https://ccomodore64.blogspot.com/2018/04/uno2iec-czyli-emulator-stacji-dyskow.html) that you acknowledged before. The seller pointed to it (as well as this GitHub), so I was convinced the parameters (incl. 57600) are okay. I will first recheck the connections more thoroughly. 
Yes, the Nano is communicating with the host app and seems okay given messages shown in uno2iec host (Windows 10, indeed). Sometimes it indicates IEC error (or two) when trying to access the virtual drive from C64, is it a hint? 
Btw. there's also port speed setting within Device Manager's COM ports section, does it matter, should it also be set to 57600? Or is it irrelevant?
I guess eventually I can try to compile and flash the code, new skill though.
Best regards,
Antoni.

Larswad, 2023-10-18:
> ![Zrzut ekranu
> (2)](https://github.com/Larswad/uno2iec/assets/114681079/24988a94-d5b1-415d-a3e6-35896fcad831)
> I couldn't replicate the error at this time.

Hi again!

Oh, great. That means you're in business now? It all works?

Remember now that my project didn't support any games that uses turbo loaders. I had plans for that, but never got around to fix it.

I did however do a port of the sd2iec project for the atmega 2560 that actually worked well. Those are somewhat more expensive than the nano though and you'd have to do some soldering of your own.

These days I can instead recommend the pi1541 project that does 100% true drive emulation. There is a bit of component ordering and soldering, but it really turns out well. I ordered a bunch of pcb's from China, "PCB way" , and got like 30 of them but only mounted and soldered like three of them. Now I don't know what to do with the rest of the unsoldered pcb's. Just lying in a box.

Ok, wish you some happy retro hours. 

Best Regards
Lars

anatolbaskak, 2023-10-19:
Hi,
Unfortunately it's not working, the Arduino side was like this from the very beginning.
I have a working 1541 and I was planning to use Uno2IEC mostly for copying from the web repositories.
I've heard about the Pi1541, maybe in the future... And I'm quite convinced the C64 community in Europe would be interested in Pi1541 boards. Through Lemon64 or some FB groups?
BRG
Antoni.

anatolbaskak, 2023-10-19:
By "the error" I meant "IEC error" message in IEC host log window.

anatolbaskak, 2023-10-24:
Okay, now it's working! The seller now sent me a manual, and the settings are even different.
[image]
Many thanks for help.

Larswad, 2023-10-24:
> ![Zrzut ekranu 2023-10-24 o 14 35

Glad to hear.

Well, I guess only he could know on what pins he chose for his specific setup.
It could have been check though by looking at the pinout of the nano and what solder connections he had done on the DIN connector. 
Have fun
Lars