Larswad/uno2iec #30: Copying from Uno2IEC to a 1541

anatolbaskak, 2023-12-07:
Hi, me again.
Since I own a working 1541, I was wondering if I can copy files onto physical diskettes from d64s using the Uno2IEC as a source drive. Given standard floppy format it *seems* possible. But I found somebody claims he checked "every dual-drive copier known to man, and they all crash" (https://www.lemon64.com/forum/viewtopic.php?t=53429).
Is there any prospect for this? Since it's possible with SD2IEC https://www.youtube.com/watch?v=khyATbkEYDI
Thanks in advance

Larswad, 2023-12-08:
Hi anatolbaskak,
I never implemented more than any of the basic functions of reading and saving through the regular commodore kernal operations.
I had ideas to implement the rest but never really ended the project in that way. There were beginnings of that code but I realised after a while how much that would demand and the time never was there.
I hope you haven't invested too much into this, it was mostly intended for me as a project to really get it working in a short time and I had hoped for more. I didn't realise the attention it later would give.
Since some still are interested I would really want to give some love to that project, but I simply don't have the time, at least not now as how it looks with kids and daytime work.
So, if you really are into arduino, you can invest in the arduino mega2560 (I think thee can be cheap clones from China) , where I have a port of the sd2iec project, based on the original sd2iec on github to work with it. That one should work cross-drive like any other sd2iec.
As an alternative, I guess there are many sd2iec builds out there to purchase or build for yourself. 
Here's my port of sd2iec for the mega2560 anyway :

https://github.com/Larswad/sd2iec_mega2560

Best Regards 
Lars

anatolbaskak, 2023-12-08:
Dear Lars,
Many thanks for prompt and empathetic reply. Don't worry, the item wasn't expensive and it fulfills its role as expected. I didn't mean to convince you to further the project, I just thought there might be an easy software solution for copying. Though I've read about it a bit and now I'm not so sure.
In any case thanks for your work and I will manage to solve my requirements, with a Pi1541 most likely.
Best regards!

Larswad, 2023-12-08:
Dear anatolbaskak,
Glad to hear you can have a way forward at least.
Came to think about a thing that might work is to load a file from uno2iec as device 9 with regular load command into memory and then do save onto the 1541, that could be device 8 or if you could even hotswap between the arduino and the 1541 before saving to/from whatever direction you want to copy. But that is file-by-file. 

Well, wish you good luck with all this, regardless.

Best Regards,
Lars